CommunityGD32Cores/ArduinoCore-GD32 #13: Add GD32F1x0 support (F130, F150, F190)

maxgerhardt, 2021-06-18:
Per title.

maxgerhardt, 2021-10-02:
With commit 91e4607f05772068d2cb975577b328956a7b5438 we now have the start of a working GD32F1x0 (GD32F190) integration that can already do GPIO, Serial and I2C.

Candas1, 2023-05-16:
Hi,

Thanks a lot for making this happen.
I tried to reach out to you with this [question](https://github.com/maxgerhardt/pio-framework-spl-with-gd32/issues/2) in the past, you probably overlooked it.
Now I believe platformio/arduino core should be the way to go.

[This](https://github.com/Candas1/Sideboard-Arduino) is a really early firmware I started to work on.

Please let me know if I can test or provide feedbakc on anything as I have many GD32F130 boards.