CommunityGD32Cores/ArduinoCore-GD32 #7: Add FreeRTOS library

maxgerhardt, 2021-06-18:
The core should have a builtin FreeRTOS library, akin to STM32FreeRTOS.

This additionally requires that the core allows hooking of the SysTick function and possibly other functions.

flute2k3, 2023-06-18:
Looking forward to testing it!

maxgerhardt, 2023-06-18:
FreeRTOS is now a core built-in library that can be used as https://github.com/CommunityGD32Cores/ArduinoCore-GD32/tree/main/libraries/FreeRTOS/examples/FreeRTOS_Blink shows. There's a default config file, but if  `GD32FreeRTOSConfig.h` exists in the `src` or `include` folder (for PlatformIO), it will be used instead. Successfully tested on a GD32F303CC.

flute2k3, 2023-06-24:
today I got chance to test the freeRTOS, I modified the codes based on the example, successfully blink 2 LEDs, impressive, thanks!

I wonder if this buildin freeRTOS does not support dynamically allocated memory? when I try to use xTaskCreate(), I got below error:
xTaskCreate' was not declared in this scope

maxgerhardt, 2023-06-24:
In the default FreeRTOS config, dynamic allocation is disabled in favor of static allocation. This prevents runtime out-of-memory situations for low-RAM microcontrollers. Irregardless, you can change the values

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/dfa7ccbef5161f0215d28d9b59aa9f24e61d07f5/libraries/FreeRTOS/examples/FreeRTOS_Blink/GD32FreeRTOSConfig.h#L51-L52

per-project to 0 or 1. Setting the latter to 1 will enable `xTaskCreate`.

flute2k3, 2023-06-24:
Hum, I will make a note on it if needed, by the way, do we have a rough standards say for example how much RAM is defined as low-RAM mcu? what about 303CC, low or ok?