CommunityGD32Cores/ArduinoCore-GD32 #9: Add CI

maxgerhardt, 2021-06-18:
The core should be built automatically on a new commit and tested against reference projects.

One could probably set up a Rasberry Pi with an actual board to act as a CI host (instead of using probably paid-for Github CI actions).

obra, 2021-06-18:
GitHub CI should be free as long as the reop is open. But that only gets us compilation tests, linting, etc.

maxgerhardt, 2021-06-18:
Ah that's nice to know, I've never worked with it yet. 

I primarily thought about using / testing https://docs.platformio.org/en/latest/integration/ci/github-actions.html and https://docs.platformio.org/en/latest/plus/pio-remote.html#technology for building projects and even remote-testing / running firmwares. Though some of it may move to https://github.com/maxgerhardt/platform-gd32 too, just like e.g. https://github.com/platformio/platform-ststm32/ does.