CommunityGD32Cores/ArduinoCore-GD32 #27: Add USB DFU upload method

maxgerhardt, 2021-06-21:
Flashing a GD32F303CC board with the popular STM32Duino USB DFU bootloader, e.g. https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/binaries/generic_boot20_pc13_fastboot.bin, shows that this exact binary works without problems on the board.

[image]

The device is being recognized via USB and opens the Maple serial as expected.

This should make it possible to add USB DFU uploads for those chips having USB but no built-in USB bootloader, that is for F30x series the 303 chips.

[image]

Important logic in PlatformIO

https://github.com/platformio/platform-ststm32/blob/38e325b59c165c954f6f69c70e0299d36f3b9225/builder/main.py#L206-L265

probably also needs LDScript changes to respect the offset.

Also needs to be implemented for the Arduino IDE.

maxgerhardt, 2021-06-21:
The first DFU upload with the pre-flashed bootloader was possible with PlatformIO, implemented in 7a0c2b19a24bd4f856f4b1aba77c54da8e215df2, 448be255d7dc61efd60ae0fd010dacb49e3dd814 and 35265fc4a404e096e2dd53109631b897614cec22.

```
CURRENT: upload_protocol = dfu
MethodWrapper(["upload"], [".pio\build\genericGD32F303CC\firmware.bin"])
Auto-detected: COM4
maple_upload COM4 2 1EAF:0003 "C:\Users\Max\Desktop\gd32_dev\gd32-pio-projects\gd32-arduino-pwm-out\.pio\build\genericGD32F303CC\firmware.bin"
maple_loader v0.1
Resetting to bootloader via DTR pulse
Searching for DFU device [1EAF:0003]...
Reset via USB Serial Failed! Did you select the right serial port?
Assuming the board is in perpetual bootloader mode and continuing to attempt dfu programming...

Found it!

Opening USB Device 0x1eaf:0x0003...
Found Runtime: [0x1eaf:0x0003] devnum=3, cfg=0, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Setting Configuration 1...
Claiming USB DFU Interface...
Setting Alternate Setting ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
Transfer Size = 0x0800
bytes_per_hash=264
Starting download: [##################################################] finished!
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
error resetting after download: usb_reset: could not reset device, win error: Ein nicht vorhandenes Gerät wurde angegeben.
Done!

Resetting USB to switch back to runtime mode
```

However, to *sensibly* do continous USB DFU uploads, the core needs to have a USB serial functionality for the Auto-reset into bootloader mode, just like the maple core does (https://github.com/stm32duino/Arduino_Core_STM32/). Since we don't have that yet, it does not make a lot of sense to work on that now (first #5). But some groundwork has been laid now.

obra, 2021-07-16:
#28 implements dfu upload to an 8k bootloader, including setting the offset of the linked code correctly for builds from within the Arduino IDE.