CommunityGD32Cores/ArduinoCore-GD32 #30: Servo code missing original copyright

michaelmargolis, 2021-08-05:
The code for the servo class and header are copied from the Arduino AVR source code but the copyright statements have been removed.  Please ensure that all copyright material used in this repository include existing copyright statements

maxgerhardt, 2021-08-05:
The servo class can impossibly be from the AVR core because it uses HardwareTimer, that doesn't exist in the AVR core.

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/2f1adfb0d687db72b0598ea3c3bbf217a0446ce5/libraries/Servo/src/Servo.cpp#L1-L4

I'll review the header more closely.

maxgerhardt, 2021-08-05:
The header seems to be mostly from 

https://github.com/arduino-libraries/Servo/blob/f3617664917ff1a46c2c3db5300d27054ba147cc/src/Servo.h#L80-L122

that is correct. Will add.

maxgerhardt, 2021-08-05:
Resolved in a68917a6c70a91eae9ae04624eb27f02fc11f475, 2b34964f3c066bf9da0c849cd96a3f86a3ae121e, ca0173f5dd71cadbf2ee01f5d76bcb417eeaddf9 and 50134d93cd1689015ce1a5e00ef7d8b07da1bd5d

closed 2021-08-05 by maxgerhardt

michaelmargolis, 2021-08-05:
Thank you for your prompt reply.

Although the interrupt code is different, the Servo class appears identical.
In this case you can add the copyright of the author of interrupt code to the original copyright.

Michael Margolis (author of the servo code)

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/2f1adfb0d687db72b0598ea3c3bbf217a0446ce5/libraries/Servo/src/Servo.cpp#L1-L4

maxgerhardt, 2021-08-05:
All done already, see above commit references.

michaelmargolis, 2021-08-05:
thanks and have fun with the project

Michael