CommunityGD32Cores/ArduinoCore-GD32 #61: implement noInterrupts/interrupts.

obra, 2022-03-19:
https://www.arduino.cc/reference/en/language/functions/interrupts/nointerrupts/ says they're part of the standard API.

We don't implement them.

It looks like it they should be implemented as wrappers around `__disableIrq` and `__enableIrq()` per https://stm32f4-discovery.net/2015/06/how-to-properly-enabledisable-interrupts-in-arm-cortex-m/ and some grepping of the GD32 code.

maxgerhardt, 2022-03-20:
I agree, we should implement it like STM32Duino does as a macro for fastest execution speed.

https://github.com/stm32duino/Arduino_Core_STM32/blob/76887a45b43c9e919db17df9e039d96bd01641d1/cores/arduino/wiring_constants.h#L85-L86

maxgerhardt, 2023-06-04:
Done in https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/a22a2e3aca8ce338f1ca8e2170fd70c53399120e quite some time ago.

closed 2023-06-04 by maxgerhardt