CommunityGD32Cores/ArduinoCore-GD32 #62: Will this ArduinoCore support GD32F350 in the future?

ldicdican, 2022-03-25:
Please add support on this m4 gd32 chip - GD32F350. Thanks.

maxgerhardt, 2022-03-25:
We already have a variant of that: https://github.com/CommunityGD32Cores/ArduinoCore-GD32/tree/main/variants/GD32350G_START

What exactly is your chip / board?

ldicdican, 2022-03-26:
My chip is GD32F350CBT6, is it compatible with GD32350G? Sorry for my being ignorant on this.

maxgerhardt, 2022-03-26:
USB is a whole different matter, this core does not yet have any USB capability. (But SPL projects like [this](https://github.com/CommunityGD32Cores/gd32-pio-projects/tree/main/gd32-spl-usb-cdc-gd32f3x0) would have). USB functionality is worked on separately as that is a whole different beast.

The "GD32350G-START" board has a GD32F350G8 chip on it. According to [GD32F350xx_Datasheet_Rev1.5](https://github.com/CommunityGD32Cores/gigadevice-firmware-and-docs/blob/main/GD32F3x0/GD32F350xx_Datasheet_Rev1.5.pdf) page 9+10 the chips have the same peripherals but different number of total GPIO pins due to their pacakge. It should be relatively easy to at least setup a blinky test for your board.

Does you board have an on-board high speed crystal (8MHz)?

ldicdican, 2022-03-27:
[image]
[image]

Actually, this chip is not on a dev board, it is on a broken vape(e-cigar) unit given to me by a friend. I was amazed that this device uses GD32 ARM M4 procie, which has LCD, buttons and some sort of sensor (?). It even has programming (SWD) pins which I labeled based on its datasheet.

I just wanna play around with this chip since I have a clone ST-Link which I use on my first bluepill board.

Thank you very much for your help and maybe I will just tinker on the GPIO pins and will ignore/keep out of my mind about the USB functionality. Btw it has also DAC, I2S, and TSI.

maxgerhardt, 2022-03-27:
Arduino core aside, you may just want to try out https://github.com/CommunityGD32Cores/gd32-pio-projects/tree/main/gd32-spl-blinky with regards to uploading and debugging.

And yes, the X1 component looks like a crystal so it should have a HSE.

ldicdican, 2022-03-27:
Thank you Max, will let you know if I have successfully make it work with ST-Link(uploading) and this gd32-spl-blinky sample.

maxgerhardt, 2023-05-07:
Fixed in https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/66300c8b0ccf97789a3f96a26f6ca05cdee61b9d

closed 2023-05-07 by maxgerhardt