CommunityGD32Cores/ArduinoCore-GD32 #64: Need help in understanding the use of "-isystem/include_path"

noeldiviney, 2022-05-01:
Hi All.
Have started to develop with GigaDevice chips.
I have followed the instructions to use the Arduino Ide and successfully compiled a the "Blink" example for the "GD32F30x MBed Series"
The following "blink-build.txt" shows the debug output of a build command.

[blink-build.txt](https://github.com/CommunityGD32Cores/ArduinoCore-GD32/files/8598375/blink-build.txt)

I would appreciate it if somebody could explain the use  of "-isystem" include paths and where I can find information on it.
I have never come across it before.
Any help much appreciated
I look forward to participating in the Community.
Best regards Noel Diviney

maxgerhardt, 2022-05-01:
@obra replaced `-I`, aka regular include flag for GCC, with `-isystem` to supress warnings (https://stackoverflow.com/a/1900578/5296568) in https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/e5ea5ab9048e979ff0a4e5a477f613af02176d2c, there's not more to that. Also see general [GCC docs](https://gcc.gnu.org/onlinedocs/gcc/Directory-Options.html).

noeldiviney, 2022-05-02:
Thank you Max for the prompt response, much appreciated.
I now have a good understanding of how -isystem works and how it is invoked in the platform.txt.
My interest is in making a embedded development environment easier to set up using Powershell scripting and using
CMake  build system and CodeLite for debugging  see https://github.com/eranif/codelite/issues/2983
It is still very much a Work In Progress.
I am actually a Hardware Development Engineer currently working on an Industrial Control Project and using the 
Arduino Framework and a GigaDevice GDF303 CM4 chip.
Many thanks once again and best regards to all.
Noel Diviney

maxgerhardt, 2022-05-02:
>My interest is in making a embedded development environment easier to set up using Powershell scripting and using
CMake build system and CodeLite for debugging see 

This problem is pretty much solved by [PlatformIO](https://platformio.org) already, which this framework supports through https://github.com/CommunityGD32Cores/platform-gd32 and the [builder script](https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/main/tools/platformio/platformio-build.py). PlatformIO downloads all the needed toolchains and can export to dozens of IDE, including CLion, VSCode, Eclipse, etc. etc., and is usable on the command line too. Have you given PlatformIO a try yet? A get started is also at https://github.com/CommunityGD32Cores/gd32-pio-projects.

noeldiviney, 2022-05-02:
Thanks for the links .
I will start some some work shortly as I have some boards and chips arriving to make it possible
Cheers Noel.

maxgerhardt, 2022-06-06:
Closing as questions answered.

closed 2022-06-06 by maxgerhardt