CommunityGD32Cores/ArduinoCore-GD32 #66: GD32F130C6 Tone and PWM doesn't work

maxgerhardt, 2022-06-02:
Per https://github.com/CommunityGD32Cores/platform-gd32/discussions/28.

No output when using provided standard code.

maxgerhardt, 2022-06-05:
Fixed a criticial issue in https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/6ba49c865388ff9dfc78f07e4e4d7f513631cb38 and it's workig on my GD32F190C6, but per discussion still no output on a real GD32F130, need to wait until realHW arrives.

djix123, 2022-06-05:
I have a GD32F130F8 to hand. Is there anything I can test?

maxgerhardt, 2022-06-05:
If you can test the sketch
```cpp
#include <Arduino.h>
PWM servo(PWM0);    // PA0

void captureCompareCallback(void);

void setup()
{
    pinMode(PB9, OUTPUT);

    // 1000Hz, 50% on-time.
    servo.setPeriodCycle(1000, 50, FORMAT_HZ);
    servo.attachInterrupt(captureCompareCallback);
    servo.start();
}

// the loop function runs over and over again forever
void loop() { }

// capture compare interrupt callback function
void captureCompareCallback(void)
{
    digitalToggle(PB9);
}
```
with `platformio.ini`
```ini
[env:genericGD32F130F8]
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
board = genericGD32F130F8
framework = arduino
```
and see whether there is output on PA0 and/or PB0, that would be great.
Make sure to use [the CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) to `pio platform update https://github.com/CommunityGD32Cores/platform-gd32.git` before though.

djix123, 2022-06-05:
Yes, that works on a GD32F130F8. Changed PB9 to PB1 since PB9 isn't available of the GD32F130F8. See attached screenshot
[image]
.

maxgerhardt, 2022-06-06:
Confirmed fixed in discussion.

closed 2022-06-06 by maxgerhardt