CommunityGD32Cores/ArduinoCore-GD32 #67: No GD32E23x MCUs in Arduino

ospilos, 2022-06-05:
Dear Sir, first of all: Many thanks for your great work!
My problem is that although there are lots of GD32XNNN micros supported in Plaformio, there are only around 5 in Arduino.
As I tend to prefer to use the Eclipse based Sloeber IDE against the Platformio, I would greatly appreciate if I can program and debug GD32E230 micros there. Unfortunately, the Sloeber plugin only accepts Arduino-style .json file.
I would geratly appreciate if you update the Arduino plugin in order to support more MCUs, namely the GD32E230K8T6, should it provided no major difficulty for you.
Best regards,
Tomáš Jíra

maxgerhardt, 2022-06-06:
We have not yet created an automation for creating the Arduino core's `board.txt`, which this Sloeber plugin probably builds on for detecting the boards. It will still take us some time to do so.

However, please just use PlatformIO to export an Eclipse project if you want to use that IDE. This [is documented](https://docs.platformio.org/en/latest/integration/ide/eclipse.html) and works perfectly fine -- there's no need to wait on anything from us.

[image]

ospilos, 2022-06-06:
Dear mister Erhard,
Many thanks for your suggestion; it really solved my issue.
Wish you good health and happines.
Tomáš

ospilos, 2022-06-06:
Working workaroud suggested.

closed 2022-06-06 by ospilos

maxgerhardt, 2022-06-06:
FYI, the recent commits (https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/48f9436d25cd3ed4b3252b4865a796d985e10969, https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/4f31b65c0f25895645fd18b206b9157db6e52545 etc.) add the auto generated boards to the board list, so all E23x boards too. In the Arduino IDE, you would need to uninstall and reinstall the GD32 core in the board manager to get the update. 

However, I would still recommend PlatformIO as the underlying build system, and not usage of the Arduino IDE or Sloeber. PlatformIO is better tested right now.

ospilos, 2022-06-08:
Dear Mr. Erhard, many thanks for your effort; my GD32E230 really works fine in the Sloeber. I am so happy now!
I do not use Platformio as the VSC seems to use the display area real estate somewhat wasteful in my opinion. Maybe it is because I do not know how to customise the UI properly, but I simply like the Eclipse's thinner borders and smaller fonts more. 
Best regards and thank you once more!
Tomáš Jíra