CommunityGD32Cores/ArduinoCore-GD32 #69: Add CAN bus libaray

CzokNorris, 2022-06-21:
Many of the GD chips support two or even three can busses. 
Would be nice to have support for that peripheral.
Should be as easy as taking the code from the STM core and moving it to the GD core.

obra, 2022-06-21:
> 
> 
> 
> 
> Many of the GD chips support two or even three can busses.
> Would be nice to have support for that peripheral.
> Should be as easy as taking the code from the STM core and moving it to the GD core.

Because of ST's restrictive licensing, we can't use code from their current cores.

maxgerhardt, 2022-06-21:
Also linked to https://github.com/CommunityGD32Cores/gd32-pio-projects/issues/7 -- I was about to tackle this.