CommunityGD32Cores/ArduinoCore-GD32 #74: Software Serial example not compiling for f130

imeshsps, 2022-08-10:
Hey,
I'm trying to use software serial with F130 but it is failing to compile in Arduino IDE with the following error.

SoftwareSerial.cpp:264:5: error: 'noInterrupts' was not declared in this scope
  264 |     noInterrupts();
SoftwareSerial.cpp:266:5: error: 'interrupts' was not declared in this scope
  266 |     interrupts();

Any help is really appreciated. Thanks in advance.

maxgerhardt, 2022-08-10:
Duplicate of https://github.com/CommunityGD32Cores/ArduinoCore-GD32/issues/61.

I've committed, https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/a22a2e3aca8ce338f1ca8e2170fd70c53399120e, can you uninstall + reinstall the ArduinoCore-GD32 in the Arduino IDE to pull the newest version and retry?

imeshsps, 2022-08-11:
Thanks for quick reply. I tried installing it following link yesterday.

[](https://raw.githubusercontent.com/CommunityGD32Cores/GD32Core-New/main/package_gd32_index.json)

Since it did not work, I tried again by installing it on a fresh virtual machine with the same link and this is the exact error I'm getting.

> C:\Users\Imesh Sachinda\AppData\Local\Arduino15\packages\GD32Community\hardware\gd32\0.0.1\libraries\SoftwareSerial\src\SoftwareSerial.cpp: In member function 'virtual void SoftwareSerial::flush()':
C:\Users\Imesh Sachinda\AppData\Local\Arduino15\packages\GD32Community\hardware\gd32\0.0.1\libraries\SoftwareSerial\src\SoftwareSerial.cpp:264:5: error: 'noInterrupts' was not declared in this scope
  264 |     noInterrupts();
      |     ^~~~~~~~~~~~
C:\Users\Imesh Sachinda\AppData\Local\Arduino15\packages\GD32Community\hardware\gd32\0.0.1\libraries\SoftwareSerial\src\SoftwareSerial.cpp:266:5: error: 'interrupts' was not declared in this scope
  266 |     interrupts();
      |     ^~~~~~~~~~
exit status 1
Error compiling for board GD32F1x0 Generic series.

Really appreciate it if you can help me with this. Is there any other way should i try to install it instead of adding above mentioned link on preferences in Arduino IDE?

maxgerhardt, 2022-08-11:
That's weird -- I'll personally test this. Maybe, but tha would be weird, it's not including the `Arduino.h` header that has this macro.

imeshsps, 2022-08-11:
Please can you look into this ? With the ST chip price going higher and harder to find, this has become an option. Thank you very much. I really appreciate the work you put into this.

maxgerhardt, 2022-08-11:
I cannot reproduce your error in PlatformIO with the latest Arduino package version.

[image]

Can you uninstall the GD32 core in the Arduino IDE again and reinstall it?

Do you see the change introduced in https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/a22a2e3aca8ce338f1ca8e2170fd70c53399120e locally? The package should be in like `C:\Users\<user>\AppData\Local\Arduino15\packages\..`.

imeshsps, 2022-08-11:
This change was not there on my installation and after I changed the file manually then it fixed the issue. Thank you very much for your help and time. I really appreciate it.

closed 2022-08-11 by imeshsps

maxgerhardt, 2022-08-11:
Okay then I'll have to double check what's up with the Arduino IDE installation routine, because the package json directly points at the master branch of this repo, reinstalling it should give  the latest version, and not the one before it.. Maybe the Arduino IDE does some package caching.