CommunityGD32Cores/ArduinoCore-GD32 #82: Support questions: upload methods

XanCraft21, 2023-04-01:
Hello, i have a question about device support for this boards package. How well is the serial upload support? Do i need to download any external software to use this feature, or is it built-in to the package? I am not able to use external software for safety reasons, so i need to know if it will work without or if there is an alternative method for serial upload. Thank you for your help.

maxgerhardt, 2023-04-02:
As per https://github.com/CommunityGD32Cores/ArduinoCore-GD32/tree/main/tools/win, this does use Gigadevice's `GD32_ISP_CLI.exe` tool, to which we don't have the source code to. This file is included in this repository so no further downloads are necessary.

I however have the feeling that for certain chips the used UART bootloader protocol might be the same as the STM32 one, so open source tools like `stm32flash_serial` as found in https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/tools/win might apply too.

Do you have a strict dependency that the serial uploader tool must be open source?

XanCraft21, 2023-04-02:
No, i do not have any strict dependencies for my project, i just need something that is already bundled with the boards package so i don’t need any extra software to download myself.

Although i really like the upload tool used in Roger Clark’s boards package from the second link above, i put it in the official STM32 boards package from STM32Duino to bypass the need to use their cube programmer. But if the GD32 package already has an upload tool built-in then i will be fine with just that.

XanCraft21, 2023-04-04:
Yes, thank you for your help, and thank you for answering my questions. I will close this now.

closed 2023-04-04 by XanCraft21