CommunityGD32Cores/ArduinoCore-GD32 #83: install package issue

MrszhuL, 2023-04-03:
OS: macOS12.6.6
IDE: vscode with arduino-cli
```shell
[Starting] Install package - GD32Community...
GD32Community:gd32:0.0.1
Invalid argument passed: Found 0 platform for reference "GD32Community:gd32@0.0.1":

[Error] Exit with code=7
```
When I add the url to Arduino settings and want to install this package some problems occurred like above. 
[image]
I have googled it but nothing found, so how to aim the problem and shoot it down?

maxgerhardt, 2023-04-03:
I have never used no registered anything with the Arduino-CLI ecosystem.

How can I reproduce what the error message? Some batch script / reproduction steps?

MrszhuL, 2023-04-03:
1. Var homebrew to install Arduino-cli
```shell
brew install arduino-cli
```
2. In vscode Arduino settings add url in the red box (ps: I can promise that this json in this IP address is as same as yours, because I just copy it to my server due to the GFW😞 )
[image]
3. then choose the board in board manager and click install. the problem maybe arise in output
[image]

maxgerhardt, 2023-04-15:
>In vscode Arduino settings 

Which extension is that exactly? Are we talking Arduino IDE 2.0 or this Microsoft extension

[image]

Since there is no "arduino-cli" extension

[image]

MrszhuL, 2023-04-15:
[image]
Yes, I have used this extension. Arduino-CLI is not an extension, it may be available from this [website](https://arduino.github.io/arduino-cli/0.19/installation/).

> Add URL in the red box

The above quote can be found in the picture below:
[image]

maxgerhardt, 2023-04-15:
While this is not fixed, can you use VSCode but deactivate the Arduino extension and install the "PlatformIO" extension and follow https://github.com/CommunityGD32Cores/gd32-pio-projects#general-usage-notes?

MrszhuL, 2023-04-15:
Thank you for your warm heart, I will follow this guide. 👍🏻

closed 2023-04-15 by MrszhuL