CommunityGD32Cores/ArduinoCore-GD32 #89: Stack size

Candas1, 2023-05-20:
Hi,

Thanks again for this project.
I am testing same arduino code on similar boards with STM32F103C8 and GD12F130C6 chips.
As I am adding more code/features, I am getting this message on the GD32 board:
[image]

Then I commented out some of the code, and I see this usage on the GD32:
[image]
And this usage on the STM32:
[image]

It looks like a fixed 2K size stack is used for GD32s ? 
It seems on the STM32 they use a minimum size for the stack and heap in the [ldscript](https://github.com/stm32duino/Arduino_Core_STM32/blob/052a49af1615fb5468ef603a0213d490d810afa2/variants/STM32F1xx/F103C8T_F103CB(T-U)/ldscript.ld#L33).

It's probably not a big problem on chips having a lot of ram, but this GD32 has only 4K of ram unfortunately.

maxgerhardt, 2023-05-20:
>It looks like a fixed 2K size stack is used for GD32s ?

Yes,

[  __stack_size = DEFINED(__stack_size) ? __stack_size : 2K;](https://github.com/CommunityGD32Cores/platform-gd32/blob/0a1c8e7fe03fb5dadd0fdb1dbb210ba428ae44ca/misc/scripts/gd32_genpinmap/ldscript.tpl#L14)

Okay, good point about low-RAM chips, though.. I'm wondering if anything useful can be done with them in the Arduino framework with such low resources :( You might need SPL to get less overhead.

Candas1, 2023-05-20:
I am actually porting this code [you know](https://github.com/EFeru/hoverboard-sideboard-hack-GD) from SPL to arduino lol
I am exploring, didn't think Arduino would have that overhead.

Candas1, 2023-05-20:
I will just use a custom ldscript in platformio.ini with parameter board_build.ldscript and a reduce stack size to something like 1K and see where I am getting.

There are hoverboard splitboards with those chips, they somehow managed to fit in motor control and reading data from an IMU.

maxgerhardt, 2023-05-23:
I've adapted the generator script to only use 512K of RAM for every chip that has 4K or less RAM.

https://github.com/CommunityGD32Cores/platform-gd32/commit/3424e663a97445312855c8b8b5ac25f55ebf0577
https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/d0cf8578ca720b778dc1acd4b56094e580d0bf2b

closed 2023-05-23 by maxgerhardt

Candas1, 2023-05-24:
Thanks a lot!!!
Much appreciated