CommunityGD32Cores/ArduinoCore-GD32 #90: issue with GD32F303C DAC1

flute2k3, 2023-06-04:
I test below codes in setup with int V_set = 2048

void setup(){
    pinMode(PA4, OUTPUT);
    pinMode(PA5, OUTPUT);
    analogWriteResolution(12);
    analogWrite(PA4, V_set);
    analogWrite(PA5, V_set);
}

it works good with PA4 (DAC0), measured voltage is 1.67V, however the voltage measured at in PA5 (DAC1) is not expected, it stays at 13mV constantly regardless of the V_set value.

I tested two boards with CC and CG chips, same issue, I assume it much be the software issue.

Thanks

maxgerhardt, 2023-06-04:
PA5 was not correctly mapped to `DAC1` indeed in the GD32F303CC_GENERIC variant. 

Fixed per https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/a1b0f8bf20e43ec78e9d90d1cef863352f05e8dc

Can you update and retest? Just [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) -> `pio pkg update -g -p gd32`.

flute2k3, 2023-06-04:
Thank you for quick action, now DAC1 is good with 1.67V output, however I get 0,63V at PA4 (DAC0). it looks to me the DAC0 and 1 can not work at the same time?

maxgerhardt, 2023-06-04:
No I think we have another bug. The `analog.cpp` function does a `dac_deinit();` when it sets up a new DAC perihperal. So when you use PA4 then PA5, the setup of PA5 causes an uncondition reset of the DAC clock domain and kills the previous config for PA4. I'll fix it.

maxgerhardt, 2023-06-04:
I pushed 

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/edc5d2b9a6b8e3a2b14c1d67a6de6db1b7fa3c5f

could you update & retest?

flute2k3, 2023-06-04:
TERRIFIC!!! now I get the same value at PA4 and PA5, amazing work maxgerhardt !

maxgerhardt, 2023-06-04:
Great! Closing.

closed 2023-06-04 by maxgerhardt