CommunityGD32Cores/ArduinoCore-GD32 #91: digital output does not work on PB3 of GD32F303C

flute2k3, 2023-06-04:
#include <Arduino.h>
#define LED PB3
void setup(){
    pinMode(LED, OUTPUT);
}

void loop(){
    digitalWrite(LED, LOW);
    delay(100);
    digitalWrite(LED, HIGH);
    delay(100);
}

Tested above codes, no plus voltage observed at PB3 pin, by another words, if I do not want to use JTAG debugging function, how can I release the PB3 PB4 pin, etc? by google searching, I found this claimed to work on stm32 does not work with gd32
afio_cfg_debug_ports(AFIO_DEBUG_NONE)

maxgerhardt, 2023-06-04:
Currently we don't yet automatically disable the JTAG / SWD pin remapping functions if we see those pins being used.

Can you try this manually as the first line in `setup()`?

```cpp
gpio_pin_remap_config(GPIO_SWJ_DISABLE_REMAP, ENABLE);
```
might need `DISABLE` instead of `ENABLE` if it doesn't work.

maxgerhardt, 2023-06-04:
Oh, actually that would completely disable JTAG + SWD (you might need to hold down reset and release it at the right time during the flashing process if you can't flash it now).

Reference manual says

[image]

So if you use SWD to flash it, you should be able to do 

```cpp
gpio_pin_remap_config(GPIO_SWJ_SWDPENABLE_REMAP, ENABLE);
```
instead, that makes PB4 available.

flute2k3, 2023-06-04:
sorry to say, but both DISABLE and ENABLE does not work

maxgerhardt, 2023-06-04:
Mhh I think it must be the **second** line, **after** you did the `pinMode()` because then the clock for the AFIO peripheral is activated and the command becomes valid.. I'll check on my hardware.

maxgerhardt, 2023-06-04:
Basically what I'm saying is
```cpp
#include <Arduino.h>
#define LED PB3
void setup()
{
    rcu_periph_clock_enable(RCU_AF);
    gpio_pin_remap_config(GPIO_SWJ_DISABLE_REMAP, ENABLE);
    pinMode(LED, OUTPUT);
}

void loop()
{
    digitalWrite(LED, LOW);
    delay(100);
    digitalWrite(LED, HIGH);
    delay(100);
}
```

flute2k3, 2023-06-04:
super, it works
but I get this error
0x2ba01477
Error: init mode failed (unable to connect to the target)
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1

maxgerhardt, 2023-06-04:
Did you upload with `GPIO_SWJ_SWDPENABLE_REMAP` or `GPIO_SWJ_DISABLE_REMAP`? You're connecting via the SWDCLK (PA14) + SWDIO (PA13) lines? 

Try also connecting the NRST of the ST-Link to the (N)RST line of the board.

Also,
>you might need to hold down reset and release it at the right time during the flashing process if you can't flash it now

maxgerhardt, 2023-06-04:
I've actually just pulled in the API for the `afio_cfg_debug_ports()` function, see

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/a178ad6ef661aff6b0b91245163ba4791f279a09

Can I ask you to update again and use the sketch
```cpp
#include <Arduino.h>
#define LED PB3
void setup()
{
    // switch to "SWD only", releasing PA15, PB3 and PB4
    // from their original JTAG/SWD function.
    afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY);
    pinMode(LED, OUTPUT);
}

void loop()
{
    digitalWrite(LED, LOW);
    delay(100);
    digitalWrite(LED, HIGH);
    delay(100);
}
```
as a test?

flute2k3, 2023-06-04:
> I've actually just pulled in the API for the `afio_cfg_debug_ports()` function, see
> 
> [a178ad6](https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/a178ad6ef661aff6b0b91245163ba4791f279a09)
> 
> Can I ask you to update again and use the sketch
> 
> ```c++
> #include <Arduino.h>
> #define LED PB3
> void setup()
> {
>     // switch to "SWD only", releasing PA15, PB3 and PB4
>     // from their original JTAG function.
>     afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY);
>     pinMode(LED, OUTPUT);
> }
> 
> void loop()
> {
>     digitalWrite(LED, LOW);
>     delay(100);
>     digitalWrite(LED, HIGH);
>     delay(100);
> }
> ```
> 
> as a test?

it works like a charm without blocking the flashing function, cool ^_^

closed 2023-06-04 by maxgerhardt