CommunityGD32Cores/ArduinoCore-GD32 #92: U8G2 does not work on GD32

flute2k3, 2023-06-04:
I tried to port the I2C OLED code from STM32duino (which is working) to GD32 but failed, the HW I2C I'm using now with GD32 is SCL(PB8) and SDA(PB9), below is the STM32duino codes associated with OLED:

#include <U8x8lib.h>
U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE); 

void setup() {
...
  u8x8.begin();
  u8x8.setPowerSave(0);
  u8x8.setFont(u8x8_font_chroma48medium8_r);
}

void loop() {
 ...

  u8x8.drawString(0,3,"ChipTmp = ");
  u8x8.setCursor(10,3);
  u8x8.print(In_Temp);

  delay(1);
}

the screen is just totally black

maxgerhardt, 2023-06-04:
I tested a SSD1306 display with the "[Adafruit SSD1306](https://github.com/adafruit/Adafruit_SSD1306)" library and that worked at the time.

Your pins are correct.

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/a178ad6ef661aff6b0b91245163ba4791f279a09/variants/GD32F303CC_GENERIC/variant.h#L96-L97

Can you run the https://playground.arduino.cc/Main/I2cScanner/ sketch to see if it detects an I2C device on the bus at all?

maxgerhardt, 2023-06-04:
Hmm actually I just plugged in my SSD1306 and it worked fine with the library.

Using
```cpp
#include <Arduino.h>
#include <SPI.h>
#include <U8x8lib.h>
U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);

void setup(void) {
    u8x8.begin();
    u8x8.setPowerSave(0);
    u8x8.setFont(u8x8_font_chroma48medium8_r);
}

void loop(void) {
    u8x8.drawString(0,3,"ChipTmp = ");
    u8x8.setCursor(10,3);
    u8x8.print("Test");
    delay(1000);
}
```
with `platformio.ini`

```
[env]
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
platform_packages = 
    framework-arduinogd32@https://github.com/CommunityGD32Cores/ArduinoCore-GD32.git
monitor_speed = 115200
lib_deps = olikraus/U8g2@^2.34.18

[env:genericGD32F303CC]
board = genericGD32F303CC
framework = arduino
```
And HW connections between my Bluepill (GD32F303CC chip):
* GND -> GND 
* 3.3V -> 3.3V
* PB8 -> SCL
* PB9 -> SDA

The code runs first try.

What's your `platformio.ini`? Can you reproduce the problem with my minimal sketch above?

[image]

maxgerhardt, 2023-06-04:
We do seem to have some.. room for improvement though, performance-wise. I've noticed that when you do a display clear and redraw, it's visibly slow. Somewhere it's hitting a timeout in the I2C code. You can lower the default timeout (`0xF0000U`) a bit when you e.g. set `build_flags = -DWIRE_I2C_FLAG_TIMEOUT=0x0F000` in the `platformio.ini`. But even at base settings, it's working for me.

maxgerhardt, 2023-06-05:
The SSD1306 OLED with 5V? Doesn't that kill it and would make all the I/O 5V?

flute2k3, 2023-06-05:
the spec says Vcc can accept 3.3V to 5V, there is a smt voltage regulator mounted on the board.

maxgerhardt, 2023-06-05:
What address does the I2C scanner report? https://github.com/CommunityGD32Cores/ArduinoCore-GD32/issues/92#issuecomment-1575668600

flute2k3, 2023-06-05:
I got these two line under serial terminal.
I2C Scanner
Scanning...

and no next

maxgerhardt, 2023-06-05:
Hm... Does my OLED maybe have 4.7K pullups to VCC on SCL and SDA? I haven't run the address scanner on my board yet, but given that the display reacts for me, I guess it should also be found. No device looks like either the device is not responding or there's an electrical problem on the I2C bus or the OLED.

flute2k3, 2023-06-05:
no lucky with 4.7k resistor pullup to 5V on both SCL and SDA.

maxgerhardt, 2023-06-05:
And when using only 3.3V?

flute2k3, 2023-06-05:
I think it is my problem, let me try it on breadboard again

maxgerhardt, 2023-06-05:
The GD32F303CC variant definition was inherited, while practically all other variant definitions were auto-generated. We don't yet have auto-generation of the Arduino variants for GD32F30x series.

The problem may be that the CG chip belongs to the "XD" (extra-density) class while the CC chip is "HD" (high-density). This actually changes some very low level register and interrupt definitions.

Could you please do the following as "hat trick": Use the `platformio.ini`.
```ini
[env:genericGD32F303CG]
platform = gd32
; use CC as a base board
board = genericGD32F303CC
; but make it seem like it's a XD chip
build_unflags = -DGD32F30X_HD
build_flags= -DGD32F30X_XD
board_build.spl_sub_series = HD
; correct flash and RAM sizes to the bigger ones (of the CG)
board_upload.maximum_size = 1048576
board_upload.maximum_ram_size= 98304
framework = arduino
lib_deps = olikraus/U8g2@^2.34.18
```

flute2k3, 2023-06-05:
use the provided platformio.ini does pass the compile, but does not solve the issue :-(

maxgerhardt, 2023-06-05:
Can you attach the debugger using the VSCode Debug -> Start Debugging menu, let the firmware run with the play button, and then pause it after some time? Does the call stack show that it's crashed in some exception handler?

maxgerhardt, 2023-06-06:
Let me create a dedicated Arduino variant for the CG chip and verify that it atleast looks and compiles good. Unfortunately I don't have a GD32F303CG chip or board to test with :(

flute2k3, 2023-06-06:
if you have a hot gun, you can replace the stm32F103C8T6 on bluepill by GD32F303CGT6, it is pin to pin replacement. the GD32F303CG itself is quite cheap, wildly available https://www.aliexpress.com/w/wholesale-gd32f303cgt6.html?catId=0&initiative_id=AS_20230606040937&origin=y&SearchText=gd32f303cgt6&spm=a2g0o.detail.1000002.0

but anway I give up the CG chip, the platformio delicately support CC chip with Arduino, and I do not actually need that much money to play with :-) let's focus our energy on CC chips first.