CommunityGD32Cores/ArduinoCore-GD32 #96: I2C on PB10 and PB11

ITstreet1, 2023-06-08:
GD32F150R8T6

Lib example I2C Scanner

I want to scan I2C device on PB10 and PB11.
This can not compile:
```
    Serial.begin(115200);
    Wire.setSDA(PB11);
    Wire.setSCL(PB10);
    Wire.begin();
    Serial.println("\nI2C Scanner");
```

'class TwoWire' has no member named 'setSDA'

maxgerhardt, 2023-06-08:
We don't provde such dynamic reassignment operations.

If you're using `board = genericGD32F150R8` then you have

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/55934353dd9cd1d22d70c78e2222e512e77a89c6/variants/GD32F130R8_GENERIC/variant.h#L111-L128

and you can see PB11 and PB10 are SDA and SCL respectively of the `Wire1` object.

You can however compile-time reconfigure the pins since they're all wrapped in `#ifndef`, meaning you can swap the pins by e.g. adding to the `platformio.ini

```ini
build_flags =
  -DPIN_WIRE_SDA=PB11
  -DPIN_WIRE_SCL=PB10
  -DPIN_WIRE1_SDA=PB7
  -DPIN_WIRE1_SCL=PB6
```

maxgerhardt, 2023-06-08:
I fully understand that this works on the Bluepill with the STM32Duino core, but we don't support [this API](https://github.com/stm32duino/Arduino_Core_STM32/blob/9fb5081a1566cb4f810b2f862a44bcc17b31de79/libraries/Wire/src/Wire.h#L82-L97) at the moment. Still, you  can compile-time reconfigure it in PlatformIO or replace `Wire` with `Wire1` in the sketch.

maxgerhardt, 2023-06-08:
That's not the only place where `Wire` is used, no? The two usages of `Wire` inside the loop should also be replaced with `Wire1`.

ITstreet1, 2023-06-08:
Oh, man... When you are tired, you should just take a break. Thank you.

closed 2023-06-08 by ITstreet1