CommunityGD32Cores/ArduinoCore-GD32 #114: gpio_interrupt.c bug

cNavsky, 2023-12-18:
Bugs in cores/arduino/gd32/gpio_interrupt.c.

1. in function "gpio_interrupt_enable"  RCU_CFGCMP clock is not enabled. As a result, adding interrupt to any pin on any port except PORTA will not work. 
Fix: add "rcu_periph_clock_enable(RCU_CFGCMP);"

2.  same function, "GD32E23x" is missing from #if defined conditions for   in "syscfg_exti_line_config". As a result, EXTI line never configured for GD32E23x chip, interrupt not working for anything except PORTA pins. 
Fix: add "GD32E23x" to#if defined conditions.

maxgerhardt, 2023-12-19:
Thanks for the report. As I see it,  `RCU_CFGCMP` exists for GD32E23x, F1x0 and F3x0, but is only enabled for the latter two, so it's not working on E23x.

maxgerhardt, 2023-12-19:
Fix attempted in https://github.com/CommunityGD32Cores/ArduinoCore-GD32/commit/047bc6f7c338f3e7619bfde433a313a2c98e6e4b, please cross-check / pull neweset version.

maxgerhardt, 2024-11-22:
Assuming fix worked.

closed 2024-11-22 by maxgerhardt