CommunityGD32Cores/ArduinoCore-GD32 #116: Is there any arduino update for the GD32E23x board now?

mayjack0312, 2024-01-06:
I tried`https://raw.githubusercontent.com/CommunityGD32Cores/GD32Core-New/main/package_gd32_index.json`
However, no support related to GD32E23x was found in the development board manager.

maxgerhardt, 2024-01-06:
The URL should be https://raw.githubusercontent.com/CommunityGD32Cores/ArduinoCore-GD32/main/package_gd32_index.json.

mayjack0312, 2024-01-06:
I imported the URL you provided
[image]
However, when searching for the development board, there was a problem that it could not be found.

maxgerhardt, 2024-01-06:
We definitely have E23x support per 

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/047bc6f7c338f3e7619bfde433a313a2c98e6e4b/boards.txt#L666-L680

The core is called "GigaDevice GD32". It might have a problem with the older version of the core still installed now. Did you remove it before switching to the new URL? If not, try and find your Arduino installation folder and delete anything GD32 from it. (`C:\Users\User\AppData\Local\Arduino15\packages`, `hardware`, ..)

mayjack0312, 2024-01-06:
> We definitely have E23x support per
>
> https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/047bc6f7c338f3e7619bfde433a313a2c98e6e4b/boards.txt#L666-L680
>
> The core is called "GigaDevice GD32". [...]

I checked all the hardware in this path and found nothing related to GD32. I have never developed this development board on an IDE before. Ask for help

maxgerhardt, 2024-01-06:
Ah. Arduino IDE 2 is special, my bad. It needs the SHA hash to be valid. Let me double check.

mayjack0312, 2024-01-06:
I checked all the contents of the subordinate folders and found the content containing gd32 in stm32duino. I have deleted it, but the development board information is still not found.
It may also be related to the version. The IDE2.2.1 version has changed a lot.

maxgerhardt, 2024-01-06:
Can you try this board URL:
```
https://raw.githubusercontent.com/CommunityGD32Cores/arduino-board-index/main/package_gd32_index.json
```
and search for "GD32" as the core.

mayjack0312, 2024-01-06:
> Can you try this board URL:
> 
> ```
> https://raw.githubusercontent.com/CommunityGD32Cores/arduino-board-index/main/package_gd32_index.json
> ```
> 
> and search for "GD32" as the core.

Okay, I tried the new link and it came out with something, but it didn’t seem to be the GD32E23x I wanted.
[image]

maxgerhardt, 2024-01-06:
The first one is still correct. We have not yet included all boards in the description.

mayjack0312, 2024-01-06:
> The first one is still correct. We have not yet included all boards in the description.

Okay, that's great, thank you for helping me solve this big problem

mayjack0312, 2024-01-06:
> The first one is still correct. We have not yet included all boards in the description.
> 
> **Edit:** I am seeing a checksum error, will try to fix shortly...

Thanks for following up on this question, this is a meaningful update

mayjack0312, 2024-01-06:
now this error
[image]

maxgerhardt, 2024-01-06:
This should be working now.

[image]

1. Remove the GD32 URL from File -> Preferences -> Board Manager URLs
2. Close VSCode
3. Delete `C:\Users\<user>\AppData\Local\Arduino15\package_gd32_index.json`
4. Delete all files in `C:\Users\<user>\AppData\Local\Arduino15\staging\packages`
5. Restart VSCode
6. Add back GD32 URL
7. Install GD32 again

mayjack0312, 2024-01-06:
> This should be working now after I made [the updates](https://github.com/CommunityGD32Cores/arduino-board-index/commits/main/).
> 
> ![grafik](https://private-user-images.githubusercontent.com/26485477/294706471-f40202f3-bbc2-4a1a-b872-550c774f516d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1NjMzOTEsIm5iZiI6MTcwNDU2MzA5MSwicGF0aCI6Ii8yNjQ4NTQ3Ny8yOTQ3MDY0NzEtZjQwMjAyZjMtYmJjMi00YTFhLWI4NzItNTUwYzc3NGY1MTZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA2VDE3NDQ1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEwYjRjODY4ZjQ5ZmU0OTgxNGY3YzlmMDNhOWQzNjc3MjA4M2YyZjc0ZGU5ZjA3MDlkNGQ1YWFhNzk0MzhiMzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KBTfClWdZdVC4f_8PH1zh5oJja5WuZQXDtBEnh_b8_8)
> 
> 1. Remove the GD32 URL from File -> Preferences -> Board Manager URLs
> 2. Close VSCode
> 3. Delete `C:\Users\<user>\AppData\Local\Arduino15\package_gd32_index.json`
> 4. Delete all files in `C:\Users\<user>\AppData\Local\Arduino15\staging\packages`
> 5. Restart VSCode
> 6. Add back GD32 URL
> 7. Install GD32 again

A very perfect solution, it can be successfully installed after this operation.
[image]
Currently it solves my problem
Thanks♪(･ω･)ﾉ

closed 2024-01-06 by mayjack0312

maxgerhardt, 2024-01-06:
Great, just remember to select the right board, chip and upload method, too.

[image]

mayjack0312, 2024-01-06:
> Great, just remember to select the right board, chip and upload method, too.
> 
> ![grafik](https://private-user-images.githubusercontent.com/26485477/294706976-79009611-45ed-43af-a56e-73ce7a72d3f0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1NjM4NzIsIm5iZiI6MTcwNDU2MzU3MiwicGF0aCI6Ii8yNjQ4NTQ3Ny8yOTQ3MDY5NzYtNzkwMDk2MTEtNDVlZC00M2FmLWE1NmUtNzNjZTdhNzJkM2YwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA2VDE3NTI1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYzNTE5MjdiYzBjMGI2MGJiZGI5NGEzNDU5YjBiOGRmYTBlNDI4NmZiZmEwNmE4ZDlhZWQxOTIwM2Y0ZWU1ZTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.615bQ-FVnmZFjA_5NzoSQnd5qjmYnY_PwDtjEWVcJJY)

Okay \(^o^)/~