CommunityGD32Cores/ArduinoCore-GD32 #119: GPIO mappings are wrong. Never get remapped to the SPL definitions. Needs complete overhaul.

bmourit, 2024-01-29:
Looking through the GPIO mapping and functions, it appears we have tried to abstract the functions away from the hardware registers, but haven't properly mapped them back to them, but rather we are using gpio_init function from the SPL, but passing it variables that are still mapped the the abstraction. Meaning, most GPIOs shouldn't be working correctly, if at all.

Since we dont have a HAL library that we are working with here, we really shouldn't abstract away much at all. There are 2 different APIs used in the SPL that need to be sorted, but we can either use the SPL header definitions directly here, or alternatively is switch methods to remap these the the proper SPL definitions to actually write to the proper registers and bits.

maxgerhardt, 2024-01-29:
Agreeing very much. This legacy thing that was carried over from the initial core needs to be reworked. I'm a bit skeptical though that it can be done completely without a PeripheralPins.h/.c, after all, the different chips all have different available pins, but using the pure GD32 SPL would expose them all equally, even ones not existing in the package.

bmourit, 2024-01-30:
Yes. I could hack something together, but I'm looking for some advice on the best/cleanest/preferred way for it to be done. I'm not sure the answer.