CommunityGD32Cores/ArduinoCore-GD32 #133: GD32F303CC I2C Scanner not working

FearL0rd, 2025-01-06:
Hi,

I am trying to use the i2C scanner with this dev board, but the scanner always returns, "No I2C devices found." But when I decided to try using the STM board, it worked. 

Any idea what could be wrong?
```

// --------------------------------------
// i2c_scanner
//
// Version 1
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    http://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//

/* Example pinmap for Bluepill I2Cs (by Testato)

 I2C-1 standard pins: PB7(sda) PB6(scl)
 Use it by "Wire" without pin declaration
  Wire.begin();

 I2C-1 alternative pins: PB9(sda) PB8(scl)
 Remap the first I2C before call begin()
  Wire.setSDA(PB9);
  Wire.setSCL(PB8);
  Wire.begin();

 I2C-2: PB11(sda) PB10(scl)
 Remap the second I2C before call begin()
  Wire.setSDA(PB11);
  Wire.setSCL(PB10);
  Wire.begin();

 If you want to use the two I2Cs simultaneously, create a new instance for the second I2C
  TwoWire Wire2(PB11,PB10);
  Wire2.begin();

*/

#include <Wire.h>

  TwoWire Wire2(PB11,PB10,1);

void setup()
{
    delay(10000);
    Serial.begin(9600);
    Wire2.begin();
    Wire.begin();
    Serial.println("\nI2C Scanner");
}

void loop()
{
    byte error, error2, address;
    int nDevices;

    Serial.println("Scanning...");

    nDevices = 0;
    for(address = 1; address < 127; address++) {
        // The i2c_scanner uses the return value of
        // the Write.endTransmisstion to see if
        // a device did acknowledge to the address.

        Wire.beginTransmission(address);
        error = Wire.endTransmission();

        if(error == 0) {
            Serial.print("I2C device found at address 0x");
            if (address < 16)
              Serial.print("0");
            Serial.println(address, HEX);

            nDevices++;
        }
        else if (error == 4) {
        Serial.print("Unknown error at address 0x");
        if (address < 16)
          Serial.print("0");
        Serial.println(address, HEX);
        }

        Wire2.beginTransmission(address);
        error2 = Wire2.endTransmission();

        if(error2 == 0) {
            Serial.print("I2C B device found at address 0x");
            if (address < 16)
            Serial.print("0");
            Serial.println(address, HEX);

            nDevices++;
        }
        else if (error2 == 4) {
        Serial.print("Unknown error at address 0x");
        if (address < 16)
          Serial.print("0");
        Serial.println(address, HEX);
        }

    }
    if (nDevices == 0)
    Serial.println("No I2C devices found");
    else
    Serial.println("done");

    delay(5000);           // wait 5 seconds for next scan
}
```

maxgerhardt, 2025-01-07:
Thanks, I will verify whether this works or not on hardware and debug it.

FearL0rd, 2025-01-07:
> Thanks, I will verify whether this works or not on hardware and debug it.

max,

I have added these to Wire.h, and now I can set the ports using ( Wire.setSDA(PB7); Wire.setSCL(PB6);), and everything is working fine.

Can you include those in the build?

```
		void setSCL(uint32_t scl)
		{
		  _i2c.scl = DIGITAL_TO_PINNAME(scl);
		};
		void setSDA(uint32_t sda)
		{
		  _i2c.sda = DIGITAL_TO_PINNAME(sda);
		};
		void setSCL(PinName scl)
		{
		  _i2c.scl = scl;
		};
		void setSDA(PinName sda)
		{
		  _i2c.sda = sda;
		};		
```

michpro, 2026-01-31:
Instead of using the standard “wire” object, you can create your own instance:
```cpp
TwoWire     i2c(PB7, PB6, 0);
```