CommunityGD32Cores/ArduinoCore-GD32 #135: 报告Error: 13 INTERNAL: Cannot install platform: installing platform GD32Community:gd32@0.0.1: testing local archive integrity: testing archive checksum: missing checksum for: ArduinoCore-GD32-main.zip

xfcai, 2025-05-25:
在用Arduino IDE安装GD32时总报告如下：
Error: 13 INTERNAL: Cannot install platform: installing platform GD32Community:gd32@0.0.1: testing local archive integrity: testing archive checksum: missing checksum for: ArduinoCore-GD32-main.zip

我单独下载了文件，安装还是不成功！
是因为安装有错误吗？有人在Arduino IDE下安装成功了吗？

谢谢

xfcai, 2025-05-25:
Failed to install platform: 'GigaDevice GD32:0.0.1'. 13 INTERNAL: Cannot install platform: installing platform GD32Community:gd32@0.0.1: testing local archive integrity: testing archive checksum: missing checksum for: ArduinoCore-GD32-main.zip

xfcai, 2025-05-25:
看到别人在使用Arduino IDE 2.0.0以上版本时也遇到了同样的问题。不知道现在解决没有？

markr1961, 2025-08-28:
Still have the same problem in Arduino IDE 2.3.6:
Error: 13 INTERNAL: Cannot install platform: installing platform GD32Community:gd32@0.0.1: testing local archive integrity: testing archive checksum: missing checksum for: ArduinoCore-GD32-main.zip