CommunityGD32Cores/ArduinoCore-GD32 #136: Consider taking USB fixes from Keyboardio's fork

obra, 2025-06-01:
Hi folks,

We did *significant* work on USB in Keyboardio's fork of this core.

https://github.com/keyboardio/ArduinoCore-GD32-Keyboardio

If someone's interested and motivated, it's worth pulling those changes back to this one.

https://github.com/keyboardio/ArduinoCore-GD32-Keyboardio/pull/13 is one such chnage.

maxgerhardt, 2025-06-01:
Are critical fixes changes in the underlying Gigadevice USB stack or on the layer above, in the Arduino core code? I have some contacts into Gigadevice if their code needs fixing.

obra, 2025-06-01:
No idea if these were dealt with in more recent vendor firmware updates. (Since things work, we're not taking them by default)

https://github.com/keyboardio/ArduinoCore-GD32-Keyboardio/commits/main/system looks like the commits we made to their firmware libraries.